pez-globo/pufferfish-software #187: `pgpiod` build and service autostart.

Tanmay06, 2020-10-07:
The new rotary encoder driver is built upon `pigpio`, which `pigpiod` daemon. The deployment process needs to build the daemon during installation process and start the daemon before the backend. This issue is related to PR #184, and issue #176.

ethanjli, 2020-10-07:
Would it work fine to install pigpiod using `sudo apt-get install pigpio` rather than downloading and building it ourselves?

Tanmay06, 2020-10-07:
I haven't tested it. I'll test it tomorrow and update it in this thread.

Tanmay06, 2020-10-08:
I guess `sudo apt-get install pigpio` is the right way to go. On `pigpio`'s repo they've mentioned that `pigpio` is currently experimental for RPi 4 , so there is a huge chance of `pigpio`s latest release breaking our deployment process. Furthermore, I've discovered that `pigpio` is pre-installed on the RPi OS version that we are using, minimizing the need to reinstall it during deployment. However, adding `sudo apt-get install pigpio` to deployment process will ensure that the daemon is present in the system irrespective of the OS.

closed 2020-10-09 by Praneet9