pez-globo/pufferfish-software #199: Add RTC support

ethanjli, 2020-10-10:
We will need to support a hardware RTC module on the Raspberry Pi so that it can keep accurate time even when the RPi is shut off and disconnected from wi-fi. This will be either an I2C or SPI device and we will need to have a kernel module driver supporting it so that it adjusts the RPi's system time.

ethanjli, 2021-04-20:
Attached are the instructions @mathengejob found to work in setting up the MAX3231 RTC. I will try to follow these instructions on my own Raspberry Pi with my own MAX3231 module and report my findings; afterwards, we can work on making a script to automatically apply these setup steps.

[Configuring RTC Time as defualt for  Raspberry PI.pdf](https://github.com/pez-globo/pufferfish-software/files/6347080/Configuring.RTC.Time.as.defualt.for.Raspberry.PI.pdf)

ethanjli, 2021-06-15:
My findings from trying to follow Job's instructions:

- It looks like I have to reboot the system after editing `/boot/config.txt` before following the remaining instructions. Initially I tried following all the instructions at once before rebooting, but then I realized that I can't initialize the time on the RTC until after I reboot once.
- After stopping at step 5, running `sudo hwclock -w`, turning off wifi, shutting down the RPi and booting it up a few minutes later, the system time was still correct (this is not true without the RTC module enabled). I did not need to perform steps 6 or 7 - it looks like step 6 is for pre-systemd versions of Raspbian.

Right now our deployment scripts are not really designed to have the system restart in between. I think what we could do is have a `setup_rtc.sh` script which runs at the start of the deployment pipeline (maybe even before `install.sh`) and can be run multiple times (detecting what it needs to do based on what's inside `/boot/config.txt`) so that we don't try to run `install.sh` multiple times - as that would slow down the deployment process with unnecessarily repeated steps.

mathengejob, 2021-06-15:
Well noted,  
Steps 1-6 will be done before the creation of the OS- image 
Step 7 is to be run during , by the end user using an interface, only once when commissioning, assuming the Wifi feature will be turned off.

PR #395 (2021-06-18): Add deployment script to set up the MAX3231 RTC
This PR fixes #199.
closed 2021-06-18 by ethanjli