pez-globo/pufferfish-software #231: Bring up Honeywell ABP sensor for FiO2 calculations

ethanjli, 2020-10-30:
In order to have correct FiO2 calculations which don't make assumptions about the ambient/current pressure, we will need to measure the ambient pressure instead of assuming it is 1 atm. This will require updating and bringing up the driver for the Honeywell ABP sensor with our protoboard, in our new architectural pattern as done for the SFM3019.

ethanjli, 2021-07-16:
Today when looking at the datasheet we realized that the Honeywell ABP sensor configured for gage pressure measurements only reports the pressure above ambient atmospheric pressure. Since the FDO2 sensor reports the partial pressure of O2 as an absolute pressure, this means that the Honeywell ABP sensor cannot be used to calculate FiO2. @hongquanli is there a different pressure sensor which we should actually be using instead of the ABP sensor for measuring the total absolute pressure in the airway so that we can calculate FiO2 as (pO2 / pTotal)? Or do we need to combine ambient atmospheric pressure (which the FDO2 sensor can report) together with the ABP's measurement of additional pressure in the airway above ambient atmospheric pressure in order to calculate pTotal as the combination of two sensor measurements?

hongquanli, 2021-07-28:
@ethanjli to keep the error for airway pressure measurement low, maybe we should continue to use that gauge pressure sensor and use combined reading for total absolute pressure?

ethanjli, 2021-07-28:
That makes sense. So what sensor should we use to measure the atmospheric pressure?

hongquanli, 2021-07-28:
There’s a BMP388 on the current version of the control board.

ethanjli, 2021-07-28:
Great, I had suspected we might need that. @rohanpurohit we can discuss the strategy for this in our next meeting, we have a few options

closed 2021-08-07 by rohanpurohit