pez-globo/pufferfish-software #232: Provide a sensible indication of FiO2 measurement when HFNC flow rate is 0

ethanjli, 2020-10-30:
Currently, when HFNC flow rate is set to 0 L/min, the ventilator is still measuring and displaying the FiO2 based on the FDO2 sensor's measurement of the (stagnant) gas inside the breathing circuit. Thus it is possible, for example, to have a setting of target FiO2 = 100% and target flow rate = 0 L/min and obtain an FiO2 measurement of 63% (Sumit has seen this in testing). This may be an edge case because it seems unlikely that a flow rate of 0 L/min would be used in practice (as opposed to pressing the "Stop Ventilation" button), but perhaps it could arise in a clinical situation. We should have a way to deal with this situation based on what would make sense for clinicians.

One approach could be to display "N/A" as the FiO2 measurement when flow rate is set to 0 L/min, to indicate that the ventilator cannot provide a reasonable FiO2 measurement at that flow rate. However, there might be other ways to deal with this situation

Sudhir-dev, 2020-10-30:
Could we set Alarm Min/Max limit for each sensor measurement. And then we could use that limit to determine if the value is above or below limits and in that case shown "N/A"?

ethanjli, 2020-11-12:
As discussed on Monday, we will hold off for now on adding any checks for alarm min/max limit; instead, for now this will just be a special case where only FiO2 is N/A (or "--") only when flow rate is 0.

closed 2020-11-17 by Sudhir-dev