pez-globo/pufferfish-software #233: Frontend is killed while changing the set values.

Tanmay06, 2020-10-30:
**Describe the bug**
The backend kills the frontend, when the user is setting any value through modal. Since the backend waits for ~1s of delay of message from the frontend before spawning the kill process, it seems like no messages from the frontend are when the modal is being operated.

**To Reproduce**
1. Start the backend and the frontend in a linux machine.
2. Open any modal
3. Keep changing the value till the chromium browser is killed.

**Expected behavior**
The backend should not kill the frontend.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device (please complete the following information):**
 - Hardware Model: [Raspberry Pi 4B]

**Software (please complete the following information):**
 - Domain: [backend, frontend]
 - Version: [0c3dfc4]

ethanjli, 2020-11-02:
This issue is related to #124: for some reasoning, opening the modal or doing other React graphics-redraw-heavy things appears to momentarily freeze all processing of the frontend (and with freeze durations which vary across different trials of the same test); when no waveform data is received for a graph 2 sec, then the graph resets itself. I am not sure how we can solve this problem - @Sudhir-dev , do you have ideas for where the performance bottleneck might be that is causing such a long freeze in the frontend?

ethanjli, 2020-11-19:
Out of curiosity, I had opened the Preact Dev Tools with redraw highlighting, and it looks like the frontend is wasting a lot redraws on things which aren't actually changing - this includes various pieces of the topbar, as well as almost everything in the dashboard. How much of the dashboard consists of nested components, vs. containers which are passed in by dependency injection?

ethanjli, 2021-02-03:
We will implement a work-around for this by limiting the rate at which the modal dialog's increment and decrement buttons may be clicked, to a rate which is low enough that the frontend can do some processing with the communication protocol in between button clicks.

closed 2021-02-16 by Sudhir-dev