pez-globo/pufferfish-software #253: Screensaver behavior

ethanjli, 2020-12-02:
**Describe the bug**
1. The Bharat Forge team reports that the screensaver does not show the current values which would be shown on the dashboard. Instead, it is only showing zero for all values.
2. The Bharat Forge team reports that they are unable to exit screensaver from the software unless they pause ventilation. I do not know if they have attempted to exit the screensaver by pressing on the Pufferfish button in the upper left corner of the screen.

**To Reproduce**
I have not yet attempted to reproduce the behaviors reported by the Bharat Forge team. We will need to first attempt to reproduce the reported behaviors, and to see if the expected behaviors are already working correctly for us.

**Expected behavior**
The screensaver should show the same values as on the dashboard, and it should be possible to exit the screensaver to the dashboard by pressing on the Pufferfish button.

**Software (please complete the following information):**
 - Domain: frontend
 - Version: unknown

ethanjli, 2020-12-03:
I have not been able to replicate either 1 or 2 on develop (8c73b599d3da4b3df112f22a7685385bec57327c).

renjipanicker, 2020-12-08:
I can confirm that it *is* possible to exit the screensaver by pressing the upper-left logo.
Perhaps it would be more intuitive to exit the screensaver by touching anywhere on the screen?

ethanjli, 2020-12-11:
It looks like Bharat Forge encountered these issues because they were running an old version of the code - we should check whether their issue is resolved by updating the software.

ethanjli, 2020-12-11:
Let's remove the pufferfish button from the left end of the topbar, and touching anywhere on the screensaver should close it.

closed 2020-12-21 by Sudhir-dev