pez-globo/pufferfish-software #254: Implement ability to set multiple buffers in MockDevice

renjipanicker, 2020-12-07:
**Is your feature request related to a problem? Please describe.**
In its current implementation, the MockDevice has single pair of buffers, one for reads and one for writes. When testing scenarios where reads/writes are performed multiple times, this single buffer-pair is insufficient.

**Describe the solution you'd like**
Implement the ability to set a array of buffers for reading and writing, so that any higher-level object, including test cases, can read/write a sequence of buffers.

ethanjli, 2021-03-12:
@renjipanicker Would it be possible for either you or @rohanpurohit to take up this issue? If so, I'll remove myself from the list of assignees and add one of you to it

renjipanicker, 2021-03-15:
I'll take it up. I should have assigned it to myself earlier.

PR #295 (2021-03-29): Implemented Multibuffer in MockI2CDevice
Closes #254 

For records-keeping:

This project is licensed under Apache License v2.0 for any software, and Solderpad Hardware License v2.1 for any hardware - do you agree that your contributions to this project will be under these licenses, too?
>> Yes.

Were any of these contributions also part of work you did for an employer or a client?
>> No

Does this work include, or is it based on, any third-party work which you did not create?
>> No
closed 2021-03-29 by renjipanicker