pez-globo/pufferfish-software #258: Display of version number/hash in frontend

ethanjli, 2020-12-11:
We need a way to display the version of software in the frontend, so that bug reports can easily be associated with a version number.

**Describe the solution you'd like**
At a minimum (and as our strongest source of truth) we should use git commit hashes; the first 6-8 chars should be sufficient. The deployment's `install.sh` script should query git for the hash of the current commit and insert it into an environment variable, which the frontend will use when building the frontend. Then the frontend should display that in the settings screen. If a git tag is associated with the current commit, the deployment's `install.sh` script should use that as the version number instead of the git commit hash.

ethanjli, 2021-04-02:
(reopening this issue because we need to add the support in the deployment script to set the environment variable added in #277)

ethanjli, 2021-04-06:
Before yarn build in https://github.com/pez-globo/pufferfish-software/blob/develop/deploy/scripts/yarn.sh, we'll need to set `export REACT_APP_VERSION="$VERSION_NUMBER"`, where VERSION_NUMBER is defined in https://github.com/pez-globo/pufferfish-software/blob/develop/deploy/scripts/helper.sh , and it could be loaded from git. When a version tag doesn't exist on the current commit, VERSION_NUMBER should be set to the commit hash. When a version tag such as v0.9.9 does exist on the current commit (i.e. it's a proper release or pre-release), that tag should be used.

rohanpurohit, 2021-04-12:
@ethanjli so I have tested this with branches where the latest commit does not have a tag and that displays the commit hash and since we recently released a tag, checked it against that commit and the tag is displayed. Although I could not get `export REACT_APP_VERSION="$VERSION_NUMBER"` to work or just `REACT_APP_VERSION="$VERSION_NUMBER"` manually and also via the script, therefore I created a `.env` file like [here](https://create-react-app.dev/docs/adding-custom-environment-variables/#adding-development-environment-variables-in-env) maybe you or @Sudhir-dev can give that a try ?

ethanjli, 2021-04-13:
This looks like a reasonable workaround for deployment. To account for any future environment variables added, I would suggest appending to .env (with `>>`) instead of overwriting.

One caveat of this approach is that if we just run in debug mode (with yarn start), we will only get a version number if we first ran the deploy script for yarn (and the frontend will display an incorrect version number if the script was run but then a different branch was switched to). If there is any way we can modify or wrap around yarn start so the version is determined at the time of the yarn command, rather than the time of the deployment script execution, that will reduce the risk of mixups or confusion with version numbers during development-phase troubleshooting. Another (not mutually exclusive) approach to prevent possible mixups is to have the yarn deployment script delete the .env file after yarn build finished.

ethanjli, 2021-04-13:
It should normally be possible to set environment variables without relying on a .env file; I've done it with the approach I described above, though that was always from bash. Did you do this testing from Windows Command Prompt? If so, the cross-env package may remove the need for the .env file.

closed 2021-04-20 by rohanpurohit