pez-globo/pufferfish-software #276: noexcept correctness on constexpr functions

ethanjli, 2021-01-15:
In Github Actions for PR #268 , clang-tidy is [complaining](https://github.com/pez-globo/pufferfish-software/pull/268/checks?check_run_id=1624837719) about exceptions possibly being thrown by functions which initialize static variables. It's not yet clear to me why clang-tidy didn't complain about these things previously; if there's a valid reason that it skipped over that code and is only catching it now that we have unit tests, then maybe the appropriate fix is to correctly use the [`noexcept` specifier](https://en.cppreference.com/w/cpp/language/noexcept_spec) in our constexpr functions.

ethanjli, 2021-03-12:
@rohanpurohit Can you see if the changes made in commit 7b31124 resolve this issue on PR #268?

rohanpurohit, 2021-03-12:
I rebased the branch on current develop, and I'm finding it hard to skim through other clang-tidy errors from test-cases,
However `clang-tidy` itself is passing now, did you mean some error from `catch2-clang-tidy` ? I think those are because of the bad function calls in test cases if I'm not wrong

ethanjli, 2021-03-12:
The clang-tidy complaint to search for in your terminal/Github Actions output is [cert-err58-cpp](https://clang.llvm.org/extra/clang-tidy/checks/cert-err58-cpp.html) - if that one still pops up then it is still a problem in the current code. The Github Actions linked to in this issue appears to have been lost now after Github activity in the past ~30 minutes (e.g. if you rebased and pushed, instead of just doing a merge), but what I remember seeing is that clang-tidy (on the catch2-clang-tidy job) complained about cert-err58-cpp when initializing `static constexpr` variables with values returned by `constexpr` functions

ethanjli, 2021-03-12:
If we can resolve or isolate all other clang-tidy errors from other test cases so that we have a branch which is only giving cert-err58-cpp errors, that could help me understand what's going on - there must be something I don't understand if clang-tidy wasn't complaining about this until Catch2 tests were defined.

rohanpurohit, 2021-03-12:
> The clang-tidy complaint to search for in your terminal/Github Actions output is [cert-err58-cpp](https://clang.llvm.org/extra/clang-tidy/checks/cert-err58-cpp.html) - if that one still pops up when you run clang-tidy on the TestCatch2 target, then it is still a problem in the current code. The Github Actions linked to in this issue appears to have been lost now after Github activity in the past ~30 minutes (e.g. if you rebased and pushed, instead of just doing a merge), but what I remember seeing is that clang-tidy (on the catch2-clang-tidy job) complained about cert-err58-cpp in header files in Core/Inc/Pufferfish which initialized `static constexpr` variables values returned by `constexpr` functions

Yes, I think those are still there, https://github.com/pez-globo/pufferfish-software/pull/268/checks?check_run_id=2092607037#step:5:46

rohanpurohit, 2021-03-12:
> If we can resolve or isolate all other clang-tidy errors from other test cases so that we have a branch which is only giving cert-err58-cpp errors, that could help me understand what's going on - [...]

Yep will do that.

rohanpurohit, 2021-03-15:
@ethanjli so i was able to isolate all the other `catch2-clang-tidy` errors, after which I picked the commit https://github.com/pez-globo/pufferfish-software/commit/7be112468e6eec7c54e80efc155ef6e45aad5009, and ran the `catch2-clang-tidy` test locally and all the ` cert-err58-cpp` errors were resolved.

ethanjli, 2021-03-15:
Great! I'll make a PR for this fix tomorrow

PR #292 (2021-03-16): Mark constexpr functions as noexcept
This PR fixes #276 by marking all `constexpr` methods and functions as `noexcept`.
closed 2021-03-16 by ethanjli