pez-globo/pufferfish-software #278: Add proper type checking in the Messages.parse method

ethanjli, 2021-02-09:
**Describe the bug**
When a uint8_t invalid type code which is not in `TaggedUnion::Tag` tries to get static casted into `TaggedUnion::Tag`, a segfault occurs.

**Expected behavior**
We could define a data structure which associates all tag uint8_t values with `TaggedUnion::Tag` enum values; if a uint8_t value isn't in the data structure, we return an error instead of trying to do a static_cast. We'll need to figure out what kind of data structure to use.

ethanjli, 2021-03-12:
I'm unable to reproduce this bug in the actual deployment setup (firmware + Python backend server, configured so that the backend server sends messages with unrecognized type code value `1` to the firmware). @rohanpurohit Can you reproduce the issue via a segfaulting Catch2 test (I remember we discussed the scenario which triggers this, and I think there's commented-out code for it in PR #269) and check whether the applying the changes from the [feature/messages-type-checking](https://github.com/pez-globo/pufferfish-software/compare/feature/messages-type-checking) branch causes the test to run successfully?

rohanpurohit, 2021-03-12:
Yes @ethanjli  the changes from [feature/messages-type-checking](https://github.com/pez-globo/pufferfish-software/compare/feature/messages-type-checking) fixes the issue.
the segfaulting Catch2 tests : 
- [type code value greater than descriptor array > 7](https://github.com/pez-globo/pufferfish-software/blob/feature/firmware-test-protocols/firmware/ventilator-controller-stm32/Core/Test/Pufferfish/Protocols/Messages.cpp#L454).
- [type code value 1](https://github.com/pez-globo/pufferfish-software/blob/feature/firmware-test-protocols/firmware/ventilator-controller-stm32/Core/Test/Pufferfish/Protocols/Messages.cpp#L465)
- [github actions](https://github.com/pez-globo/pufferfish-software/runs/2093837079)

ethanjli, 2021-03-12:
Great, I'll create a PR tomorrow. I'm surprised the first one is segfaulting, I thought the parse method specifically checked for that case - I'll take a closer look at what's going on

rohanpurohit, 2021-03-12:
Oh ya I think this was the edge case where it's equal to the array size. 

https://github.com/pez-globo/pufferfish-software/blob/develop/firmware/ventilator-controller-stm32/Core/Inc/Pufferfish/Protocols/Messages.tpp#L62
should be 
`if (type >= pb_protobuf_descriptors.size()) {`

ethanjli, 2021-03-12:
Right, I remember discussing that but I don't think the fix has been made yet. Maybe that was the final GIthub Issues item I had lost and forgotten about hahaha. I'll include that fix in this PR.

PR #291 (2021-03-13): Improve type code checking in firmware's Messages implementation
This PR resolves #278 by:

- Defining a new `Util::EnumValues` variadic class template which provides a recursive constexpr function to check whether the integer provided as input is the underlying value of one of the enum values provided as a template parameter to the `Util::EnumValues` template specialization. To use this, specialize `Util::EnumValues` with an enum type and the enum values from that enum type (e.g. `using MyEnumValues = Util::EnumValues<MyEnum, MyEnum::first_value, MyEnum::second_value, MyEnum::third_value>`),  [... cut]
closed 2021-03-13 by ethanjli