pez-globo/pufferfish-software #282: Vector::copy_from and calling code should detect when copy is truncated to prevent overflow

ethanjli, 2021-02-23:
`Util::Vector::copy_from` returns silently, even if it couldn't copy all source data as requested because its internal buffer filled up in the middle as a result of the copy. It should return an `IndexStatus` indicating whether an out-of-bounds error was averted - i.e. it should copy what it can, and use `IndexStatus::out_of_bounds` to indicate if it was requested to copy data which would've been out of bounds of its internal buffer. Anything which uses Vector::copy_from should also check and handle its return code. For example, `Protocols::CRCElement::parse` currently silently truncates data if its payload buffer is shorter than necessary to hold all data from the input buffer, because that is the behavior of `Util::Vector::copy_from`.

For CRC elements, it may also be possible to enforce this through the compiler: we could add a static assertion into `CRCElement::parse` and/or `CRCElementReceiver::transform` comparing `input_size` against `PayloadBuffer::max_size()`, and into `CRCElement::write` and/or `CRCElementSender::transform` comparing `output_size` against `PayloadBuffer::max_size()`. These static assertions make `CRCElement` less flexible in ways we don't care about, so it could be good to add this extra level of checking if the compiler allows it.

ethanjli, 2021-02-23:
@rohanpurohit I think this was one of those issues I had lost when my tabs crashed!

rohanpurohit, 2021-02-24:
oh ok! will work on it.

PR #286 (2021-03-09): firmware: Vector::copy_from: detect when copy is truncated
This PR resolves #282 by adding return codes to Vector::copy_from, in it's current state it only copies source data to internal buffer if the source data size is less than or equal to the internal buffer size, if the size of the source data is greater than internal buffer size it returns an `IndexStatus::out_of_bounds` error code.

Also Wherever copy_from is used:
`CRCElement::parse`, `CRCElement::write`, `Datagram::parse`, `Datagram::write`, 
- The copy_from method is executed before other members are updated. 
for ex: In CRCElement:: [... cut]
closed 2021-03-09 by rohanpurohit