pez-globo/pufferfish-software #325: Landing Screen should only allow starting

ethanjli, 2021-04-21:
Right now the landing screen allows the user to bypass the "start" button by clicking on the sidebar; we should remove the sidebar. Additionally, the "start" button always works, even if the backend is not available - in which case pressing the "start" button takes the user to the Quick Start screen and immediately triggers the "backend connection lost!" alarm. We should prevent the "start" button from allowing this. The simple approach we discussed was to disable the "start" button and make it say something like "Loading..." or "Connecting..." instead of "Start" while the backend is unavailable.

(this issue was created from https://github.com/pez-globo/pufferfish-software/pull/313#issuecomment-818406959 )

ethanjli, 2021-04-30:
See https://github.com/pez-globo/pufferfish-software/blob/7a046d81b0b995a59f5a33e060fccddf60670a85/frontend/src/modules/app/OverlayScreen.tsx#L45 for inspiration and/or reuse

rohanpurohit, 2021-05-01:
@ethanjli can you confirm if 8128529902bbf1cad89d2737e5edbbb3dbfa0db9 is the behaviour we would need and are there any regressions from it?
changes:
- landing page will dispatch a backend lost event, start button will be disabled and show 'loading'
- if backend is lost in quickstart or anywhere else. start button will be disabled and show 'loading'

closed 2021-05-11 by rohanpurohit